bancika/diy-layout-creator #795: V4.37 will not open on my Mac

assaf1101, 2023-10-04:
I get an error massage - "DIYLC 2" cannot be opened because the developer cannot be verified.
macOS cannot verify that this app is free from malware.

MacBook Pro
Sonoma 14.0

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* One request or bug report per issue, please. No bundling.
* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2023-10-04:
you just need to allow the application to run. Check how it's done for your
version of OS, it's different for each one.

assaf1101, 2023-10-04:
[image]
sure, there is no such option - only to move to trash, as you can see in the screen shot.

[image]

bancika, 2023-10-04:
https://it.nmu.edu/docs/allowing-third-party-applications-install-macbook#:~:text=Select%20%22Security%20%26%20Privacy%22%20from,the%20left%20of%20%22Anywhere.%22

assaf1101, 2023-10-04:
Worked! Thanks!

closed 2023-10-04 by assaf1101