bancika/diy-layout-creator #804: Highlighted Connected Areas tool connects over a cut in a vero strip

visuttaja, 2023-12-04:
I wonder if it was possible to make Highlighted Connected Areas tool take those cuts in vero strips in to consideration.
That would be super helpful to way to debug the composition.
Currently the tool jumps over the cuts creating new connections which are not there.
That kind of makes me more confused and I think I do better job just trying to check it out hundreds of times like I have always done.
It would create a huge time saver if the algo was able to terminate the tracing at the cut.
I insert a screeen shot where I tried to follow the positive power rail of the step up board output.
[image]

visuttaja, 2023-12-04:
Ok. Here goes...
[HP_amp_given_dual_rail.txt](https://github.com/bancika/diy-layout-creator/files/13549941/HP_amp_given_dual_rail.txt)

visuttaja, 2023-12-04:
For some reason it started to work. I try to recreate the situtation and find the simplest possible steps to catch the giltch.
But I guarantee this was not the first time I was discouraged by the feature so there really must be something going on.

visuttaja, 2023-12-04:
I cannot create the problem in simplest form using brand new vero board and inserting the cut. HCA works as expected.
BUT. I can recreate the situation again with the original 9v HP-amp which had two opamps. 
I started my new simpler version of HP-amp by using the old one and the problem carried on with the copy.
Now here must be the needed clue for you. I try to keep it intact so that when opened and then just selected HCA and strip it keeps the problem alive.
[HP_Amp_with_opa_voltage_HCA_tool.txt](https://github.com/bancika/diy-layout-creator/files/13550587/HP_Amp_with_opa_voltage_HCA_tool.txt)

visuttaja, 2023-12-04:
Here again is the image of what I see with the old file (HP_Amp_with_opa_voltage_HCA_tool.diy) in the first place.
[image]

visuttaja, 2023-12-04:
I sorry to grab your time. I see there is some problem above with cuts not being in the middle of the strip but in the first pic they are. I think this is my bad here.

visuttaja, 2023-12-04:
I think I accidentally found the simplest way to reproduce the issue! There is 13 steps.
1.Start DYILC from disk as sole instance.
2.Insert a veroboard
3.Insert a trace cut on a strip. 
4.Turn on HCA. 
5.Press on the strip with cut. HCA works as expected.
5.Turn off the HCA.(this should wipe the connections off....it doesn't which could be part of the issue)
6.Select the board and start editing it from working area(right click -> Edit Selection).
7.Change Coordinate Origin of the board to any other value. Say Top Left. 
8.Turn on HCA.
9.Press on the strip with cut. Now you should get:
[image]
10. Turn off the HCA.
11. Move the cut.
12. Turn on HCA. 
13. HCA works again.

visuttaja, 2023-12-05:
So current workaround is to turn off HCA and move any of the elements on vero when editing is again possible. Then after reengaging HCA it works as expected.

bancika, 2025-02-25:
Fixed for 4.41

closed 2025-02-25 by bancika