bancika/diy-layout-creator #807: How to get folded caps in DIYL

stillampd, 2024-01-08:
Using V4.37
In Passive/Electrolytic Caps (radial)
mine does not show "folded caps" mine shows "no variants"
How do I get the folded caps?

[image]

bancika, 2024-01-09:
double click on a capacitor on the drawing, not in the menu

stillampd, 2024-01-14:
Here is what others are getting.
[image]

stillampd, 2024-01-14:
What I want to do is get a radial cap on its side
[image]

M0JXD, 2024-04-06:
Hi,

I don't have the option variant option either:
[image]

However, if you double click on the component you can open a menu where you can select the folded look:

[image]

^ Double Click

[image]

^ Select the box

[image]

^ Folded cap.

closed 2025-02-09 by bancika