bancika/diy-layout-creator #811: Feature suggestion: Export silk screen

perette, 2024-03-02:
It's possible to make a silkscreen from a circuit by adjusting image and colors, but it requires a bunch of manual steps.  It would be nice if there was a built-in way to do this, similar to File->Trace Mask->Export.  The silk screen export would:

- export all layers except Chassis.
- Override trace width to 0.5mm or so.
- Override component outlines to black.
- Override component colors to white.
- Override all text, both component and misc, to black.

The resulting image is suitable for home-built boards, printing onto Press-N-Peel and ironing it onto the top-side of the board; or for running through ImageToPcb to make a gerber for a silk screen layer for having a board manufactured.

negout, 2024-08-13:
I have just started using diylc and made a PCB layout. I was going to get the PCB made by an online manufacturer that requires a gerber file. What is exactly is a gerber file. I have not seen a way to generate in diylc. I have exported it to pdf file but that is all.

perette, 2024-08-13:
@negout There is a separately-developed utility [ImageToPCB[(https://imagetopcb.com/) which can accept image files (exported PNGS from DIYLC) and generate Gerbers.  See also [these instructions](https://deviousfish.com/making-pcbs.html#sect_3.3) or [this video](https://www.youtube.com/watch?v=Rnj_Ri5jxR4) for an overview of the process.

bancika, 2025-02-18:
@negout @perette gerber export functionality is currently under development and will be out probably early march. I'm hoping that it will include the silk screen layer as well. Currently handles outline, drill, solder mask, copper. If it succeeds for gerber I guess I could do it for PDF and PNG as well. 

@perette question about your request - shouldn't silk screen be monochromatic? You are suggesting mixing white and black which would be problematic for fabrication I think.

perette, 2025-02-18:
Yes, silk screen should be monochromatic.  Currently exporting the silk screen to a PNG does so in color, requiring color adjustments to make it monochromatic.