bancika/diy-layout-creator #813: Boards Lib

UMDONI, 2024-03-14:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* One request or bug report per issue, please. No bundling.
* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

UMDONI, 2024-03-14:
When clicking on "Vero Board" in the Boards Directory to select..... DIY Layout App  hangs-up and I have to close it with windows 7 task manager in order ..to continue where I left off.

bancika, 2024-12-17:
Hi @UMDONI can you please upload diylc logs so I can investigate?

bancika, 2025-02-10:
Closing due to inactivity. Please add more info if you want to reopen