bancika/diy-layout-creator #815: Current number of push pull DPST switch limitations is 4. Need it to be at least 5 or 6.

astromiester1, 2024-03-29:
Can you update the code to include more switches. Currently, I need 5 or 6 push pull DPSTs or 4 with the ability to add 2 more toggles.

The software guitar analyzer function says number of switches max has been reached anything after 4 DPSTs.

Thanks
Dale

bancika, 2024-03-29:
This is configurable, you can add parameter to the application

-Dorg.diylc.maxSwitchCombinations=X

By default it's set to 64.

It will take longer to process and the report will be gigantic as it
outputs switching configuration for each of the combinations of the
switch...be aware.

astromiester1, 2024-03-29:
Do I add this inside the application somewhere or do I have to recompile the code to make a new build for this option?

Thanks Dale

bancika, 2024-03-29:
Which OS are you using?

astromiester1, 2024-03-29:
I have win10 pro

astromiester1, 2024-04-02:
I have win10 pro. Whats next? Do I need to recompile using Eclipse Java or can I make the change in the current build at the application user level?

Thanks
Dale

astromiester1, 2024-04-02:
How exactly do I use this new feature in the program that you added?  I found this was added to the program back in 4.36 4-27-2023.

 <li>[New Feature] Configurable max switch combination count for circuit analysis (using org.diylc.maxSwitchCombinations argument)</li>

Thanks Dale

closed 2024-04-16 by bancika