bancika/diy-layout-creator #818: How and where do I use the -Dorg.diylc.maxSwitchCombinations=X feature using lastest build?

astromiester1, 2024-04-06:
Can you please tell me how to use this feature you added to the latest build... Where do I place this text? In the project somewhere or on the command line? I am using Win 10 Pro.

Thanks
Dale

M0JXD, 2024-04-06:
Hi Dale,

I don’t know if this will work (I use Linux), but try downloading the universal build. (The diylc-4.37.0.zip on the releases page) and adding the “-Dorg.diylc.maxSwitchCombinations=X” to the diylc.l4j.ini file, which seems to hold some command line arguments for Windows.

For the *nix universal build (Linux/macOS), you can add the “-Dorg.diylc.maxSwitchCombinations=X” to the run.sh file.

astromiester1, 2024-04-07:
Hi Jamie, thanks for the reply... I did look in the DIYLC progs folder and did not find an ini file to alter... I wish the developer would just tell me where it goes when he finds some time.. Very Frustrating.. .

Thanks
Dale

astromiester1, 2024-04-07:
Ok...I downloaded the 437 source and noticed the ini file in the download folder... I edited it with the switching command limit I wanted and ran the diylc.exe in that folder and it worked running from the source code folder... I wouldn't mind an installed version that works as well. So, I tried recompiling the code using eclipse java IDE with the switching line in the diylc core folder where a config.properties file resides. Anyway, I could build diylc-core, diylc-library but not the diylc.swing folder as it failed with the following error "An internal error occurred during: "Launching DIYLCStarter".
Cannot invoke "org.eclipse.jdt.launching.IVMInstall.getLibraryLocations()" because "install" is null"

Bottom line although, I would like to be able to compile and run the source code in Eclipse Java IDE platform, I do have a workaround that got me going... Thanks for all your help Jamie... Hope, I can continue to solve this problem as I really lilke this application.  Thanks Dale

bancika, 2024-04-16:
I understand your frustration, however, please bare in mind that

* In most cases I don't have answers in my head that I can quickly provide, it takes time/effort to check
* Family and my regular work takes precedence and I have very little time to dedicate to DIYLC these days
* The Issues page is for actual issues and this is more of a 'customer support' request, which I officially do not offer
* I've been working on this app totally free for the last 18 years! The little time I can dedicate to it usually goes towards improvements and new features

That said, try creating/updating **diylc.l4j.ini** file with this line

`-Dorg.diylc.maxSwitchCombinations=256`

astromiester1, 2024-04-16:
Ok..I ended up doing that very thing a few weeks ago..ThanksDaleSent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone

closed 2024-04-16 by bancika