bancika/diy-layout-creator #819: Intermittent unresponsive clicking?

eddielovesyou, 2024-04-22:
Running Sonoma 14.4.1 on an Intel Macbook Pro
Latest DIYLC 4.37.0

Just downloaded today, thank you for the amazing application! 
I'm trying to create a guitar wiring diagram and the mouse click will often not register any action. I might have to click a component 3-4 times to select it, or it might work on the first click. Same with clicking open space to deselect. Right-clicking will always bring up the context menu but has the same intermittent behavior when it comes to selecting the component I've right-clicked. Selecting components in the right sidebar Project Explorer always works as intended.
I've tried changing to Config > Toolbox > Tabbed Toolbar instead of the default Searchable Tree as recommended in an older issue thread, but that didn't help. 
Thanks for any suggestions!

Log attached
[log-2024-04-22 10.10.17.txt](https://github.com/bancika/diy-layout-creator/files/15067291/log-2024-04-22.10.10.17.txt)

pablomuri, 2024-08-17:
I have the same problem

bancika, 2025-02-11:
Hello. Is this still an issue with >4.39 versions? @pablomuri @eddieziegler-multco

pablomuri, 2025-02-11:
Hi @bancika. Yes, it's still happening. The first click that doesn't work is still happening to me. How can I dump the logs?

bancika, 2025-02-11:
Are you using intel or arm version?
Logs can be found in ~/diylc/logs, you can access them from the Help menu
in DIYLC

pablomuri, 2025-02-11:
I am using ARM version in a macbook pro m1 pro MacOS 15.3. I can't see any evident error.

[log-2025-02-11 11.52.05.txt](https://github.com/user-attachments/files/18748831/log-2025-02-11.11.52.05.txt)

bancika, 2025-02-11:
Nothing suspicious. I tried it on M2 PRO and it does happen intermittently.
Every 5-10 attempts, the click doesn't register. I'll investigate

bancika, 2025-02-20:
Fixed for 4.40

closed 2025-02-20 by bancika