bancika/diy-layout-creator #821: Slide switch

And-E-Roo, 2024-04-27:
Hi,
I love DIYLC, it's helped me immensely with my guitar wiring projects. 
I am unable able to get the slide switches to function  correctly, though. I assume , since the manual doesn't have schematics, that the middle pin on a SPDT and DPDT switch would be the common or output, and the outer ( top and bottom) pin or pins be input or travelers. This does not work, and unless I am missing something, doesn't seem to work in any configuration I've tried. I even started on a new project with just a pickup, output jack and slide switch and both SPDT and DPDT don't seem to function correctly. Am I doing something wrong in the wiring  here? I took screenshot of a wiring diagram that has 5 of the six poles junctioned together, yet still does not work unless I place the pickup lead wire on the same pole as the connection to the jack. Any feedback would be greatly appreciated. I'm looking forward to posting some of my own diagrams in the near future, thanks again for an amazing product!
[image]

And-E-Roo, 2024-05-12:
Anybody hear them crickets?

M0JXD, 2024-05-17:
Sorry to leave you hanging - please remember that DIYLC is a free project that @bancika has been working on in his free time for nearly 20 years. There isn't any customer support like commercial software.

That said he tries to help and I do too when I can, although I don't see how this is a DIYLC problem? To be honest I'm not entirely following, what are you trying to achieve with the switch?

By manual, do you mean DIYLC's manual or the manual/datasheet for the switch? Slide switches are pretty much universally wired as so:

[image]

Switches with more "poles" are activated in parallel but are not electrically connected.

Could you post/attach a diagram with the problem? As an example that might help, a very simple switching diagram for a telecaster using a slide switch could look as follows:

[image]

One side would choose the neck pup, the other the bridge.

I hope this helps!

waldorfcrane, 2024-07-05:
I'm having the same problem with any sliding switch over SPDT
This works
[image]

This doesn't
[image]

Works
[image]

Doesn't
[image]

Here's a weird test that's hard to understand, but what I think it shows is that the tracks are behaving like a different kind of switch
[image]

bancika, 2025-02-13:
this is a bug in the switching logic for double and triple slide switches

closed 2025-03-19 by bancika

And-E-Roo, 2025-03-19:
I got a new one for you. I downloaded the updated version about a week or two ago, and when you try to undo a move,(control z keys on Mac) it undoes several moves, not just the last current one. I still think your product is pretty awesome though. Thanks for putting it out there.

Sent using the mobile mail app

On 3/19/25 at 6:06 AM, Bane Stojkovic wrote:

bancika left a comment (bancika/diy-layout-creator#821)

fixed for one of the upcoming versions

bancika, 2025-03-19:
> pretty awesome though. Thanks for putting it out there. <br> <br> Sent
> using the mobile mail app <br> <br> On 3/19/25 at 6:06 AM, Bane Stojkovic
> <br> <br> fixed for one of the upcoming versions <br> <br> -- <br> Reply to

there was a bug in a recent version that made it ignore all mouse move
operation for undo, so it would jump all the way back to the last action
prior to moving the selection. Maybe it's that? If so, it should be fixed i
5.0.0