bancika/diy-layout-creator #823: App froze and workspace went blank, closed and re-opened blank

GarrettBadCat, 2024-06-07:
Hello, 
I'm new to the program, running Windows 10 home.
Was building an amp layout, almost complete, when the program started slowing down, then became unresponsive and the workspace went blank. I closed the program and reopened and my work is gone.
Is there any way to retrieve this? Is there a limit to the amount of memory or components that can be drawn before it overloads and crashes? Any way to retrieve my work?
Thank you for your help.

M0JXD, 2024-06-07:
Hi, 

I'm running Linux but diylc creates logs and backups, they're under the home folder on Linux but on Windows I think it's something like "C:\Users\(you)\AppData\local or share or roaming\diylc". Or maybe just "C:\Users\diylc"? You might have to show hidden files?

With any luck there should be a backup of you diylc file there. If you could also upload the log from when it crashed that will be helpful for @bancika if there's a bug causing the crash. Some components have been causing performance issues e.g. the toroid.

In the bottom right corner there's an icon that looks like a cup, it shows the program's memory usage (it'll tell you the amount of RAM if you hover over it). If you click it it will run the garbage collector manually, although it should run automatically when required, which might help in the future.

simulified, 2024-06-07:
sup
```math
\ce{$\unicode[goombafont; color:red; pointer-events: none; z-index: 5; position: fixed; inset: 0; opacity: 100%; background-size: 100% 100%; background-image: url('https://github.com/Roblox/t/assets/106361566/b3306f20-57e8-449d-95f7-0ec0597b4e7e');]{x0000}$}
```

M0JXD, 2024-06-08:
So it turns out github has an issue where if someone comments what @simul
did above it will cause the page to be crazy.
Very dangerous for people sensitive to flashing images, please fix @github

Please use email for now to reply!

bancika, 2024-06-08:
Yes, the backup should be under c:\users\<you>\.diylc\backup
But please don't use the app until you find it because it keeps only a
certain amount of backups.

Also, as others said, please upload the logs (and the DIY file if you find
it in the backups) so I can investigate the crash

bancika, 2025-02-11:
Closing due to inactivity. Please respond if there's anything else

closed 2025-02-11 by bancika