bancika/diy-layout-creator #831: Cannot Open or Save Projects

smallbeerfruit, 2024-09-11:
Hi 
I am unable to save any projects and also I can only open projects from the recent files drop down - not from the file menu.
I can open these files from the relevant documents folder on my laptop though.

There are no error messages when I try to save or open a file - nothing happens at all. I have downloaded the latest version (v4.37.0?). This suddenly started happening with no obvious changes to my laptop etc.

Any ideas???

M0JXD, 2024-09-11:
Which OS are you using?

I know that the Flatpak version for Linux has issues with accessing files due to permissions, if you are running linux I recommend the universal build.

smallbeerfruit, 2024-09-11:
Hey thanks for replying. It's Windows 11 (version 23H2 apparently..)

smallbeerfruit, 2024-09-17:
Interesting...I loaded the software onto my old Windows 10 laptop and it works fine, so I guess it's something to do with the last Windows 11 update. Do you know where I would find the permissions, if that's what it is? (I'm awful with computers.....)

M0JXD, 2024-09-17:
Hmm - I've got a dual boot running Windows 11 (which is 23H2 as well) and I haven't any issues saving and opening projects. I reinstalled to double check. If you haven't already try re-installing; "off and on again" is the usual Windows trick 😅

I've had issues in the past due to OneDrive (Which Windows seems to default to much to my dismay). Try navigating to "C:\Users\you\Documents" to save stuff instead of the "C:\Users\you\OneDrive\Documents" folder that is default in the file manager.

The issues thing is only for Flatpaks on Linux so I don't think that's the issue

bancika, 2024-09-20:
@smallbeerfruit could you please try to upload a few log files created around the time you experienced the issue? You can find them in help menu in the app and then 'access user files'

bancika, 2025-02-20:
@smallbeerfruit when you say "nothing happens at all." does that refer to the moment you click on "Open" in DIYLC menu or it lets you pick a file and then nothing happens once you pick it?

smallbeerfruit, 2025-02-20:
Well since then, it will open files and I can amend them, but I can't then
save under a new file name. When I try nothing happens. As in there is no
reaction at all, as if I hadn't tried to save.

bancika, 2025-02-20:
Can you please try once to save and send me the latest log? It lets you pick a filename, click on Save and then nothing, right?

smallbeerfruit, 2025-02-20:
Sure. I'll do this at the weekend. Busy prepping for a gig tomorrow night
at the mo!

Many thanks for your help so far.

Martin

smallbeerfruit, 2025-02-24:
Hey, Good News!

I downloaded the new (January version) and everything works fine now.

Thanks so much for all your help.

Martin

bancika, 2025-02-24:
great, thanks for the update!

closed 2025-02-24 by bancika