bancika/diy-layout-creator #832: Pasting a building block or copy pasting a grouped object error

EL34xyz, 2024-09-19:
Bancika told me to post this here so it gets on the fix it list.
He noted that this was the issue
**"It seems that z-order of the lugs gets out of whack was an issue"**

I was trying to show a guy on Facebook how he can make his own dual pot from scratch
Then you group everything and save it as a building block
Also how to bring forward and send backward items so everything displays correctly

That all works fine, but if I copy and paste the dual pot or paste a building block, the second row of terminals does not show up
Even though they are brought to the front and the blue area is brought to the back.

The image on the left shows the original dual pot I drew and then made a building block
The image on the right is the building block I pasted
The same thing happens when I copy and paste the one on the left

[image]

bancika, 2024-09-20:
could you please paste the building block as it is saved in your config.xml? I'm trying to establish if it's the problem in saving the block or recalling it.
Thanks!

EL34xyz, 2024-09-21:
Thanks Bane.
I copied the DualPot entry section from the config.xml and posted it below.

      <entry>
        <string>DualPot</string>
        <list>
          <diylc.shapes.Rectangle>
            <name>RECT1</name>
            <alpha>127</alpha>
            <value></value>
            <controlPoints>
              <point x="0.21875" y="0.59375"/>
              <point x="1.09375" y="1.0625"/>
            </controlPoints>
            <firstPoint x="0.21875" y="0.59375"/>
            <secondPoint x="1.09375" y="1.0625"/>
            <color hex="ccffff"/>
            <borderColor hex="000000"/>
            <borderThickness value="0.001" unit="in"/>
            <edgeRadius value="0.125" unit="in"/>
          </diylc.shapes.Rectangle>
          <diylc.shapes.Rectangle>
            <name>RECT10</name>
            <alpha>127</alpha>
            <value></value>
            <controlPoints>
              <point x="0.25" y="0.96875"/>
              <point x="1.0625" y="1.03125"/>
            </controlPoints>
            <firstPoint x="0.25" y="0.96875"/>
            <secondPoint x="1.0625" y="1.03125"/>
            <color hex="ffcc33"/>
            <borderColor hex="000000"/>
            <borderThickness value="0.2" unit="mm"/>
            <edgeRadius value="0.0" unit="mm"/>
          </diylc.shapes.Rectangle>
          <diylc.connectivity.SolderPad>
            <name>Pad1</name>
            <size value="0.1875" unit="in"/>
            <color hex="cccccc"/>
            <point x="0.375" y="0.8125"/>
            <type>ROUND</type>
            <holeSize value="0.1" unit="in"/>
            <layer>_1</layer>
          </diylc.connectivity.SolderPad>
          <diylc.connectivity.SolderPad>
            <name>Pad2</name>
            <size value="0.1875" unit="in"/>
            <color hex="cccccc"/>
            <point x="0.65625" y="0.8125"/>
            <type>ROUND</type>
            <holeSize value="0.1" unit="in"/>
            <layer>_1</layer>
          </diylc.connectivity.SolderPad>
          <diylc.connectivity.SolderPad>
            <name>Pad3</name>
            <size value="0.1875" unit="in"/>
            <color hex="cccccc"/>
            <point x="0.9375" y="0.8125"/>
            <type>ROUND</type>
            <holeSize value="0.1" unit="in"/>
            <layer>_1</layer>
          </diylc.connectivity.SolderPad>
          <diylc.connectivity.SolderPad>
            <name>Pad4</name>
            <size value="0.1875" unit="in"/>
            <color hex="cccccc"/>
            <point x="0.375" y="0.46875"/>
            <type>ROUND</type>
            <holeSize value="0.1" unit="in"/>
[... 102 lines cut ...]
            </controlPoints>
            <firstPoint x="0.625" y="0.875"/>
            <secondPoint x="0.6875" y="1.03125"/>
            <color hex="cccccc"/>
            <borderColor hex="cccccc"/>
            <borderThickness value="0.2" unit="mm"/>
            <edgeRadius value="0.0" unit="mm"/>
          </diylc.shapes.Rectangle>
          <diylc.shapes.Rectangle>
            <name>RECT9</name>
            <alpha>127</alpha>
            <value></value>
            <controlPoints>
              <point x="0.90625" y="0.875"/>
              <point x="0.96875" y="1.03125"/>
            </controlPoints>
            <firstPoint x="0.90625" y="0.875"/>
            <secondPoint x="0.96875" y="1.03125"/>
            <color hex="cccccc"/>
            <borderColor hex="cccccc"/>
            <borderThickness value="0.2" unit="mm"/>
            <edgeRadius value="0.0" unit="mm"/>
          </diylc.shapes.Rectangle>
        </list>
      </entry>