bancika/diy-layout-creator #834: Telecaster neck Missing? - 2 and 3 leads?

ClayfordG, 2024-10-03:
I can't seem to locate tele neck pickups. Since you have a 4 way tele, I'm likely missing it somewhere and also the 3 wire version as well

M0JXD, 2024-10-03:
You can make the single coil look like a Tele neck pickup quite easily:

ClayfordG, 2024-10-03:
Yeah for two wires that would work. As three three wires on a tele neck pickup are needed needed for 4 way switching (kind of why I assumed I was missing it somewhere super obvious)  I guess I could add a connection point on the two wire pup, but it floats on the wrong layer.  Using a four wire pup would be super confusing for beginners looking at wiring diagrams, and obviously the circuit tester would fail.

bancika, 2024-12-16:
you can jumper the two middle pins together and tap a 3rd lead from either one of them...that's essentially what it is