bancika/diy-layout-creator #836: Component request: screened wire / coax

JezCunningham, 2024-10-15:
Hi - loving the tool - creating a wiring diagram to incorporate an overdrive pedal circuit inside a Stratocaster body and i want to show which wires need to be screened.  You have hook-up wire and twisted pair - a coax would be very useful!
Thanks
Jez

bancika, 2024-12-16:
Hi. Screened wire can be easily created using a combination of several regular wires of varying gauges - one for the core, one for the shield and a thick one to cover everything. That way it can be saved as a building block and recalled as many times as needed.

Typically I avoid adding components that can be easily covered with existing ones.

Here's an example, blue leads are shielded

[image]

JezCunningham, 2024-12-16:
That’s clever! Thanks!
Jez

closed 2025-02-09 by bancika