bancika/diy-layout-creator #837: Cannot launch DIYLC after upgrade to MacOS Sequoia 15.1

Nerdvana2, 2024-11-02:
DIYLC no longer launches on my Mac in the normal way; I *think* this started with the OS upgrade to Sequoia 15.1.

Double-clicking on a .diy file or on the DIYLC app just produces a Finder pop-up saying 'The application "Finder" does not have permission to open "(null)".'

However I can launch DIYLC from the zsh shell, with the command "/Applications/DIYLC.app/Contents/MacOS/JavaAppLauncher"

Any thoughts or suggestions? Is it just me??? :(

bancika, 2024-11-04:
Does the app appear under Privacy and Security in system settings?

I tried running DIYLC on sequoia and it worked after I allowed it from
there.

bancika, 2024-11-05:
To fix this you have to delete and create a signature for the OS to open
it.  Run the following command in a terminal window:

sudo codesign --remove-signature /Applications/DIYLC.app

sudo codesign --force --deep --sign - /Applications/DIYLC.app

After you have run them you will need to allow the app in privacy and
security and then it should work again

Nerdvana2, 2024-11-05:
Thanks - after the two codesign operations the application is launching as expected.  I did not need to change anything in "Privacy & Security".

Any ideas about why this happened? I get the impression that it was a problem with my machine, not a general DIYLC/Sequoia 15.1 issue.

Thanks again; Sean

closed 2024-11-05 by Nerdvana2