bancika/diy-layout-creator #838: SLOW on 4.37.0 VERSION of DIYLC

indrasung, 2024-11-04:
Hello forum
I just upgrade DIYLC from 4.33.0 to 4.37.0
Placing components , then rotate or move now become so much SLOWER than before. 
I go back to 4.33.0.

Does any body also experience this ?

Thanks

bancika, 2024-12-13:
which OS are you using?

bancika, 2024-12-13:
Also, do you have any example file?

M0JXD, 2024-12-13:
This is probably the toroidal inductor issue still as that fix is yet to be released.

indrasung, 2024-12-14:
Hello Bancika
My OS is 64 bit Windows 7.
No I do not have any example files.

bancika, 2024-12-15:
It's crucial for me to have at least some clue, otherwise it's impossible
to run any investigation. Does it happen with any file you open? Does any
particular component type trigger bad performance? Give me something to
work with :)

bancika, 2025-02-20:
Closing as I'm unable to replicate. @indrasung if you have any further information I could use feel free to reopen

closed 2025-02-20 by bancika