bancika/diy-layout-creator #843: org.freedesktop.Platform//22.08 is end-of-life

rsholmes, 2024-11-15:
Attempting to install in Linux Mint via

```
flatpak install flathub com.diy_fever.DIYLayoutCreator
```

I get

```
Info: org.freedesktop.Platform//22.08 is end-of-life, with reason:
   org.freedesktop.Platform 22.08 is no longer receiving fixes and security updates. Please update to a supported runtime version.
```

(I could use the Universal build, but that is very inconvenient due to not being integrated into the desktop environment and having to run from the terminal after navigating to the installation directory.)

bancika, 2024-12-13:
Try this
flatpak install --user https://dl.flathub.org/build-repo/137250/com.diy_fever.DIYLayoutCreator.flatpakref

I merged a PR to upgrade to 24.08, so it should build shortly

M0JXD, 2024-12-13:
FYI, I use the universal build with Linux Mint and create my own desktop entry.
Presuming you're using mint cinnamon, right click on the "start menu" button and select "Edit Menu". The provided editor is pretty straightforward.

I put the unzipped diylc folder in a ~/Applications folder (increasingly required for me due to portable apps and appimages), modify run.sh as per my comment in #721 and create the desktop entry as mentioned, copying the name and descriptions from the provided .desktop file.

closed 2024-12-16 by bancika