bancika/diy-layout-creator #844: Loading/Saving in Flatpak

ChrisOfBristol, 2024-11-18:
I've just tried to load an old diagram, it didn't work as I couldn't see it, so I tried saving a file then reloading it. It did not appear in the directory. I tried "Save as" with the same result.

I'm on version 4.37.0 on Fedora-41. As it's a Flatpak I found the logs as expected in ~/.var/app/com.diy_fever.DIYLayoutCreator/diylc/logs

[log-2024-11-18 09.59.01.txt](https://github.com/user-attachments/files/17798426/log-2024-11-18.09.59.01.txt)
[log-2024-11-18 09.54.33.txt](https://github.com/user-attachments/files/17798427/log-2024-11-18.09.54.33.txt)
[log-2024-11-18 09.51.59.txt](https://github.com/user-attachments/files/17798428/log-2024-11-18.09.51.59.txt)
[log-2024-11-18 09.48.19.txt](https://github.com/user-attachments/files/17798429/log-2024-11-18.09.48.19.txt)
[log-2024-11-18 09.46.41.txt](https://github.com/user-attachments/files/17798430/log-2024-11-18.09.46.41.txt)

ChrisOfBristol, 2024-11-20:
A workaround is to install Flatseal then go to DIYLC > Filesystem and click on "All user files".

bancika, 2025-02-10:
Added this information to the FAQ, thanks!

closed 2025-02-10 by bancika