bancika/diy-layout-creator #845: Feature Request: Add ability to insert rows & columns in veroboard

XLOOX, 2024-11-23:
Love DIYLC .  Love your other work too. have used DIYLCt for about 3y on ~15 PCBS. Bought a couple of coffees for you in that period.

I like making compact vero layouts but how may times have I  given myself insufficient realestate and get to the last few bits and find I havent the space ??    ... Loads....

One thing I would DEARLY love is the ability to insert another vero row or column at a specific place in a veroboard layout so that everything remains linked to its current hole but their connects stretch . 

Now I just have to start from scratch :(

cheers

Andrew

bancika, 2024-11-23:
Hi Andrew. Thanks for the kind words and the coffee :) As for the
feature...I will consider it, but as a workaround, doesn't it make sense to
select all components to the side of row/column you want to insert and just
nudge them for one grid space to the right or down? Sure, this still
requires manual intervention for components that span across the row/column
you are adding.

XLOOX, 2024-11-24:
Hi Bane,
 Thanks for replying so quick.
Yes  I do nudging at times but I find that often I have left some bits behind ( prob ones that span the break) & if I dont notice then I stuff up my circuit and mainly find out when I print & trace by hand over the vero & compare to the circuit diagram.

Speaking of which - Is it possible for you to add in another layer   - one that the user can simply draw/highlight/write on?  I personally would find it invaluable when double checking the layout vs the circuit diagram to be able to highlight as I go along following a trace to each component.

I would then know which bits of the circuit  I have checked and made sure i have got the connections right, put arrows on current flow, highlight the earth strips .etc. I could then hide/ the whole layer before printing.

I already do some  markup using the shapes & text fields but to print  have to  delete stuff that I would rather just hide

Not sure if I need to submit as another feature req.

Thanks for considering anyway.

regards

Andrew

bancika, 2024-12-16:
please file a separate request, so it doesn't get lost. It's an interesting idea