bancika/diy-layout-creator #846: Analyze Guitar Diagrams error when deleting single component

StateOfEpicicity, 2024-12-04:
Please run the Analyze Guitar Diagrams command for this file, then delete C3 and run again.  After deletion, an error will pop up reading "org.diylc.netlist.TreeException: Could not find anything connected to a jack tip terminal."  The diagram would still work without C3, but for some reason the analyzer doesn't recognize this.
[HH 5-Way VV Push Poll Series Parallel Resonant Peak Shifts.zip](https://github.com/user-attachments/files/18010148/HH.5-Way.VV.Push.Poll.Series.Parallel.Resonant.Peak.Shifts.zip)
[log-2024-12-04 08.56.24.txt](https://github.com/user-attachments/files/18010169/log-2024-12-04.08.56.24.txt)

bancika, 2024-12-16:
the thing is that after deleting c3, in one of the switching configurations that's nothing connected to the output and the process fails. I should relax the condition to allow it to work if there's at least one switch configuration that will produce sound, but not all.

StateOfEpicicity, 2024-12-16:
The reason this one stumped me is I have worked on other diagrams in which I run the analysis before I'm done, and for the switch positions I haven't yet connected, it will display something like "no pickups connected" for those positions.  So there are some diagrams in which it will work when not all switch positions are connected.

The other confusing thing is the wording of the error message itself: "Could not find anything connected to a jack tip terminal."  It's no problem if the analyzer was meant only to work when all positions produce sound, but the wording in that case should be something like: "Not all positions produce sound."

This also overlooks one potential usage for a switch position with no connector, to act as a kill switch!

StateOfEpicicity, 2024-12-16:
I'm wondering if the diagrams in which the analyzer will say that a switch position has no pickups selected are ones in which I've included another type of switch altogether, like a blade and a rotary or a blade and a mini toggle.  I'll need to look at that when I get a chance.

Thanks for looking at this!

bancika, 2025-02-09:
Changed the logic to analyze each position individually and allow having some that don't work

[image]

closed 2025-02-09 by bancika

StateOfEpicicity, 2025-02-09:
That's so awesome.  Thank you so much for doing this!