bancika/diy-layout-creator #847: Save only shows an arbitrary subset of folders in home

php4fan, 2024-12-15:
To reproduce on Linux:

- click Ctrl+S or go to File-> Save As
- By default this will open your home folder (`/home/username` e.g. `/home/teo`). If not, navigate to your home folder

Expected behavior
=
I should see all the folders that exist within my home folder, and should be able to open them and navigate to wherever I want to save my project

Observed Behavior
=
It only shows this arbitrary subset of subfolders:
[image]

My home directory contains plenty of other non-hidden subfolders, and I actually want to save my project within one of those.

php4fan, 2024-12-15:
The non-flatpak version (AUR package "diylc") doesn't have the issue.

This might be a duplicate of https://github.com/bancika/diy-layout-creator/issues/844

bancika, 2024-12-16:
yeah it's a flatpak issue that doesn't let DIYLC see those folders. I don't know enough about flatpack to tell what exactly causes it. I suggest installing flatseal and trying to manage flatpack permissions from there

https://flathub.org/apps/com.github.tchx84.Flatseal

bancika, 2025-02-09:
duplicate of #844

closed 2025-02-09 by bancika