bancika/diy-layout-creator #850: Permission issue starting DIYLC in Mac OS 15.1.1 Sequoia

mramonlopez, 2024-12-18:
If you install the last version of DITLC on a Mac with OS 15 Sequoia and try to run it, you could get the error 

> The application "Finder" does
> not have permission to open
> "(null)".

It's a problem with the signing of the .app file. You can fix it in your computer running:

`codesign --remove-signature /Applications/DIYLC.app
`

and then:

`codesign --force --deep --sign - /Applications/DIYLC.app
`

Source: https://forums.macrumors.com/threads/finder-does-not-have-permission-to-open-null.2441374/?post=33535657#post-33535657

LudfisterSound, 2025-01-04:
This worked for me to get rid of the dock null errors but I did still have to attempt to open the app and then go to Privacy & Security in Setting and allow the app to run under the Security section all the way at the bottom.

bancika, 2025-01-04:
Yes, that's due to Apple's requirement for everyone to distribute apps
through apple store. I plan to offer the app there but it will probably
cost a few dollars to download it there because it will cost me some cash
for apple developer account and it will require more effort to publish
there too. Stay tuned, I plan to do it soon

MarsManiac, 2025-01-18:
Also an issue for me.  PowerBook, Intel, OSX 15.2.  Thanks for prompt attention.   NRN

mbvanemby, 2025-01-19:
The solution above worked well for me. Thanks!

MarsManiac, 2025-01-19:
Those commands worked without error.  Then I get 

'You can’t open the application “DIYLC.app” because it may be damaged or incomplete.'

Is there a workaround for this?

bancika, 2025-01-19:
You must enable it as a 3rd party app, link here
https://github.com/bancika/diy-layout-creator

closed 2025-02-10 by bancika