bancika/diy-layout-creator #853: "Ctrl +" zooms in but "Ctrl -" doesn't zoom out (Spanish keyboard)

php4fan, 2025-01-06:
version 4.37.0

I think the title says it all.

bancika, 2025-02-19:
is spanish "-" key different in any way? Maybe it sends a different code?

php4fan, 2025-02-19:
Yes, the keyboard layout is different

[image]

The key code seems to be 189, I don't know which one it is in an US English keyboard but I'm pretty sure it's different.

bancika, 2025-02-19:
hm it's 189 on my mac and both + and - work

php4fan, 2025-02-19:
> The key code seems to be 189

Hm, that's what I got from Javascript in a browser but I'm not sure if that's reliable.

`showkey` in a terminal says it's 53.

This is on Manjaro Linux with KDE and Wayland.

bancika, 2025-02-20:
try this...extract this zip and then run

java -jar swingTest.jar

then press "-" in the text field and you'll see the code

[swingTest.jar.zip](https://github.com/user-attachments/files/18884996/swingTest.jar.zip)

php4fan, 2025-02-20:
Apparently the version of Java I have installed is not recent enough

```
Error: LinkageError occurred while loading main class Main
        java.lang.UnsupportedClassVersionError: Main has been compiled by a more recent version of the Java Runtime (class file version 61.0), this version of the Java Runtime only recognizes class file versions up to 55.0
```

I'm not sure why I have such an ancient version, I keep everything up to date with Pamac (Manjaro Linux)

php4fan, 2025-02-20:
Ok, seems to be 45:

[image]

Why javascript in the browser says one thing, `showkey` another, and Java yet another, is beyond me.

bancika, 2025-02-20:
Ok I think I found the issue..it's in the code. Will fix