bancika/diy-layout-creator #856: Printing bypasses dialog box

michaeleschuler, 2025-01-11:
When I select the print option in Windows 10,  Windows opens up the default print dialogue box but the software prints immediately. The resulting printout is cropped so that the entire layout doesn't appear on the page, even though the printout is also not fitting to the page (that it, is small, but also cropped).

bancika, 2025-01-16:
Managed to replicate this bug...will fix when I have a chance

michaeleschuler, 2025-01-16:
Thanks!

⁣Regards,

Michael Schuler
Berger+Schuler Editing Partners
206-214-6128

LinkedIn: michael-e-schuler
Website: BergerPlusSchuler.com

Get BlueMail for Android ​

bancika, 2025-01-16:
An easy workaround is to save to pdf and then print using other apps

bancika, 2025-02-09:
fixed

closed 2025-02-09 by bancika