bancika/diy-layout-creator #857: Looking for Guidance on Shape Dimensions

martiro7, 2025-01-13:
First, thank-you for a great tool! I use it on all my amplifier builds.

I use MacOS Sequoia 15.1.1 with DIYLC 4.37.0 because I am unable to get around security issues with the latest version at the moment. (I'm looking into JDK).

**Issue #459 Set Dimensions of Boards or Shapes** shows that this feature was added back in version 4.01 yet I am unable to access this capability after dragging a rectangle, line, or ellipse onto the canvas. Can you provide guidance on how to reach the shape properties for setting dimensions?

My current workaround on ellipses is to drag an electrolytic capacitor onto the canvas, which allows me to set the diameter. Thank-you for your time.

bancika, 2025-01-13:
Double click and change sizing mode to explicit dimensions or something
like that. Then you can type in width and height values

martiro7, 2025-01-13:
Thanks for the quick response.

When I double-click, this is all I get in the dialog:

[PastedGraphic-1.png]  Or with rectangles [PastedGraphic-2.png]

I don’t see anything about dimensions.

Thanks,
Rob Martin

bancika, 2025-02-09:
Added support to rectangles and ellipses as well..

closed 2025-02-09 by bancika

bancika, 2025-02-09:
> That’s phenomenal news. I’ll give it a try as soon as I’m back at my bench
> and download the latest.
>
> Thank-you!
> Sent from my iPhone

It hasn't been released yet

martiro7, 2025-02-09:
Got it. Thanks