bancika/diy-layout-creator #858: Does not have permission to open "(null)."

martiro7, 2025-01-15:
Running DIYLC on macOs Sequoia 15.1.1 generates the following error (see attached) which is only circumvented by Show Package Contents and launching JavaAppLauncher.
[image]

mbvanemby, 2025-01-16:
Same issue here. I'm also running 15.1.1 on an Intel Macbook Pro.

bancika, 2025-01-16:
see this https://github.com/bancika/diy-layout-creator/issues/850

I plan to publish to mac store but apple is not very cooperative and it will cost me some money and time...so it will cost a few bucks to download it from there but there will be no hassle.

closed 2025-01-16 by bancika

martiro7, 2025-01-16:
I know I would gladly pay