bancika/diy-layout-creator #861: Increase maximum undo history depth

php4fan, 2025-01-25:
On a project I'm working on, I needed to undo a few dozens steps, and I quickly reached a point where "undo" wouldn't go further. This was a ridiculously small number of steps.

I understand that you have to put a limit or else one will eventually run out of memory, but I'm pretty sure you can increase the current limit by a factor 100 or 1000 without getting even close to any practical issue (unless something is tremendously inefficient).

bancika, 2025-02-01:
Current limit is 32. You can try to change it by adding a command line argument

-Dorg.diylc.appframework.undo.UndoHandler.MAX_STACK_SIZE=1000

bancika, 2025-02-01:
if that doesn't work try

-Dorg.diylc.appframework.undo.UndoHandler<?>.MAX_STACK_SIZE=1000

bancika, 2025-02-09:
Increased to 128 by default

closed 2025-02-09 by bancika