bancika/diy-layout-creator #863: Show/hide layers makes you lose selection. Impossible to select multiple components in fornt of / behind one another

php4fan, 2025-01-28:
Say I have a couple of components next to each other, and a trace that is behind them, completely covered by them and hence inaccessible to click and select.

Now I want to select one or both of the components, plus the trace behind.

So what I do is I click on one component to select it, then I hit Ctrl+4 to hide the components, and then I CTRL+click on the trace. And then I hit Ctrl+4 again to show the components again.

But every time I hit Ctrl+4, I lose my selection, so the trick above does not work.

And not only that: if I select a few things, then do Ctrl+4 back and forth just to look at things without doing nothing, I lose the selection.

And there doesn't seem to exist any way to select both a component and a trace that is behind it unaccessible to clicks.

I also tried this: first select the component, then attempt to right-click on the trace (but the component is on front so I'm right-clicking on it) and in the context menu do "select -> trace <wahatever>". However, this will select the trace alone and de-select the component. (I also tried holding the Ctrl key while doing this) So there's no way to accomplish the selection I want, unless there's an even more convoluted and undiscoverable way.

I think both the methods above should work. 

1) Show/hide shouldn't affect the selection. Now one may argue: but if I select something and then hide a layer and end up with hidden selected objects,  I will not know that those are selected and this is dangerous. I may move them unknowingly, and only find out afterwards when I unhide the layer, or worse, delete them (thinking I'm only deleting the stuff that I see) and only find out hours later when I realize something is missing. That's true. So I would propose this: when you hide a layer and there are selected components in that layer, maybe show a popup with a warning and/or asking for confirmation, and then, when the layer is hidden, show a red, empty outline around the selected invisible items, so you always know at every moment that you have selected components that are hidden.

2) First of all, when you right-click on something and show the context menu, currently the very act of right-clicking immediately selects the object and forgets the current selection, before you even do anything with the context menu, and even if you then close the context menu without doing anything, the selection is lost anyway. In other words, a right-click behaves like a left click and _then_ opens the context menu.  That's just plain wrong and a bug, regardless of anything else. The action of right-clicking and  opening a context menu should have no consequences at all; only when you do some of the actions in the context menu, should that have consequences. But then, suppose you fix that, the problem comes when when you do "select -> [item]". Now, I propose that if you already have a selection, this new selection gets added to, rather than replacing, the current selection. There's no downside to that: if you don't want that, you just click outside to clear the selection before you do right-click->select->[item]. Or, if you want, you can add a popup prompt when you do "select->[item]" (if anything is already selected) asking whether to add to or replace the current selection (personally I think this is unnecessary). Or, maybe holding ctrl while doing "select -> [item]" could have the effect of adding the new selection to the current one, just like when you hold ctrl while clicking. This is perhaps the most consistent option but may be a little difficult to discover.

BTW showing and hiding layers currently count as actions that are part of the undo history and I guess that complicates this further, but I think that's wrong on its own and I'll report that as a separate issue.

bancika, 2025-01-28:
> and in the context menu do "select -> trace ". However, this will select

" showing and hiding layers currently count as actions that are part of the
undo history"

this information is stored in DIY file so it has to be part of the undo
history.

bancika, 2025-01-28:
> The fact that it's stored in the DIY file (and that's good, it would be a
> mistake not to save it in the file, I do expect it to remember what was
> hidden and what was not when I open a file I saved - although incidentally
> it wouldn't be a huge deal if it didn't) doesn't mean it *has to* be part
> of the undo history. It should be pretty trivial to store the current state
> in the file without having it in the undo history; I can't even imagine how
> that would be difficult, but if it is, there's probably a design issue
> somewhere else. Either way, it certainly doesn't have to be that way.

It's a conscious decision I made early on to tie undo stack to actions that
make changes to the file. Otherwise you wouldn't be able to go back. I
don't know a single piece of software where you cannot undo action that
changes the contents of the file

php4fan, 2025-01-28:
I have opened a separate issue for the undo/redo history thing, I've replied there: https://github.com/bancika/diy-layout-creator/issues/864

php4fan, 2025-02-11:
> Stuff like "That's just plain wrong and a bug, regardless of anything else." is jumping to conclusion and a bit aggressive. That's how many applications work - Finder on mac, Windows explorer, Corel. Are they all wrong and dumb?

I was wrong about that indeed. Most applications do select on right click. Because I was trying to work around the issue by using the "right-click->select..." feature and that got in my way, I didn't realize that's actually the norm. I don't see why you should take that as an aggression though (and I never said "dumb"). I simply made a wrong statement, thank you for correcting me on that.

bancika, 2025-03-19:
@php4fan you can also multi-select components in the project explorer sidebar. It's handy for cases like this