bancika/diy-layout-creator #864: Show / hide layers is part of the undo history

php4fan, 2025-01-28:
I understand that what layers are hidden and what layers are not is stored in the file, so when you load a file you saved, each layer's visibility is remembered as it was when you saved the file. That's good.

However, the action of hiding and showing layers is currently also part of the undo history and that's not desirable.

Besides the minor annoyance of having to undo and redo several additional unnecessary steps when you go back and forth (and besides the fact that these counts towards the maximum number of remembered steps which is very very limited as per #861), this has much graver consequences. Suppose I do the following:
- I undo many changes
- I hide and then show some layers, just to see how things look
- then I intended to go "back to the future", i.e. redo everything. But I can't, because by merely showing and hiding layers I made new changes and hence lost everything in the undo history after the point where I stopped to show and hide.

bancika, 2025-01-28:
> Replying to #863 (comment)
>
> It's a conscious decision I made early on to tie undo stack to actions
> that make changes to the file.
>
> Ok, then I it may be time for some refactoring.
>
> Otherwise you wouldn't be able to go back
>
> Again, not true.
>
> I don't know a single piece of software where you cannot undo action that
> changes the contents of the file
>
> That seems hard to believe, but anyway that's irrelevant. If that were
> true, then you have the opportunity to be the first person in the history
> of software to implement a nice and useful feature that is so incredibly
> easy to implement, and be an example for all other software to copy.
> The point in time in which you show and hide layers with respect to when
> you do all other actions is completely irrelevant, so there's no need to
> store it in order. I assume you're storing every action that makes up the
> content of the file in a sequence (otherwise I wouldn't see why this is an
> issue in the first place). Like:
>
> 1. place this component there
> 2. place this other component there
> 3. show  a layer
> 4. hide a layer
> 5. delete that component
> 6. move that other component
> 7. hide a layer
>
> You can just keep the show/hide layer operations in a separate list (which
> BTW doesn't even need to be a list, you just need the set of current
> visible layers or that of hidden layers or a map from layer number to
> boolean) which is not undoable and which is stored e.g. at the end of the
> file, and applied at the end when loaded.
>
> By the way, this hard tying of the undo history to the actions that make
> changes to the file sounds like something that could have a couple of nice
> side effects: (1) undo history could be preserved across saving and
> reloading the file, and (2) it could be "infinite", i.e. as long as the
> actual entire sequence of actions that built the contents. And yet we have
> neither of those, so it cannot be *that* deeply tied, unless you decided
> on purpose to not have those features.

I intentionally didn't do what you proposed because it would infinitely
increase the file size as well. Diy projects can easily grow to several mb
if you start adding images. Maintaining unlimited undo history with every
single action in the file would make it grow exponentially.

php4fan, 2025-01-28:
Ok that makes perfect sense. But then, the undo history is already not "one" with the structure of the contents. All the more reason it shouldn't be too difficult to separate out the layers visibility.

M0JXD, 2025-01-28:
I think a better solution for power users who need "unlimited undo" might be to use something external like Git to track their DIYLC project. Other Git features like branches could be pretty handy too. Perhaps a plugin could be developed to provide some integration.