bancika/diy-layout-creator #865: Transform selection -> rotate is broken

php4fan, 2025-01-28:
I selected this:

[image]

I did "Edit -> Transform selection -> rotate clockwise", and I got this:

[image]

The "mini toggle switch" (blue thingie) is not where it should be.

php4fan, 2025-01-28:
And here's something way worse:

This:

[image]

rotated clockwise becomes this:

[image]

Not only the mini toggle switch has moved, but also, **everything is now misaligned with respect to the grid**.

This cannot be aligned with a board that was not rotated together with the components, I can't draw traces connecting the rotated components, and I can't fix the mini toggle switch, because it's now connected to traces it wasn't connected to before rotation, and if I cut it and paste it again, it will get aligned to the grid and misaligned to the other rotated components.

The whole thing is unusable.

php4fan, 2025-01-28:
[TEST_BUG_ROTATE.diy.zip](https://github.com/user-attachments/files/18580541/TEST_BUG_ROTATE.diy.zip)

This reproduces the misalignment issue. It's just two traces:

[image]

If you move the one on the right just one grid unit up, it will rotate correctly.

No idea whether the mini toggle switch issue is related.

bancika, 2025-03-17:
@php4fan misalignment issue is fixed for 5.1.0 and is unrelated with the other one.

bancika, 2025-03-17:
Also, you can always use ctrl and alt to unstick a component temporarily and move it to the desired location without affecting anything else

bancika, 2025-03-18:
fix the switch issue as well. Please report if you notice any other issues, they need to be handled separately.

closed 2025-03-18 by bancika