bancika/diy-layout-creator #868: Auto-wrap label removes empty lines

php4fan, 2025-02-02:
Put this text in an auto-wrap label:

```
Lorem ipsum dolor sit amet.

Lorem ipsum dolor sit amet.

Lorem ipsum dolor sit amet.
```

This is the result:

[image]

The empty rows are completely ignored.

I'm not sure if that is supposed to be a feature, but I see no advantage in that.

I'm forced to choose between using an "Abc label", which does honor empty lines but doesn't do the auto wrapping, and Auto-wrap label which does (pretty bad) auto-wrap but doesn't allow me to add empty lines, e.g. between paragraphs.

bancika, 2025-02-10:
Yeah, there's no auto-wrap feature in Java so I had to improvise it.  You'll have to live with it for the time being. You can create a label-per-paragraph as an easy workaround

bancika, 2026-01-02:
Fixed for 5.9.0 @php4fan