bancika/diy-layout-creator #872: Fonts of the menu and windows is very big, unworkable

janneman001, 2025-02-09:
I tried to use the program in Linux OpenSuse 15.5

When i run the program the fonts are huge.  The screen resolution is: 3840x2160

[image]

I tried on another computer with the same OS and the fonts were fine. The screen resolution here is: 1920x1080

How can i make the software workable on higher resolution?

The program looks very promising. I would like to use it. Hence my question.

bancika, 2025-02-09:
Can you try adding jvm argument to run.sh file? It should go together with
the rest of -D arguments

-Dsun.java2d.dpiaware=false

janneman001, 2025-02-10:
thanks,

I added it to run.sh. Unfortunately the same result.

I noted some warnings in the console however. Maybe this gives some clue.

WARNING: package com.apple.eawt.event not in java.desktop
WARNING: package com.apple.eio not in java.desktop

Also an error trace:

11:07:09,361 ERROR [AWT-EventQueue-0] MainFrame - Background task execution failed
java.util.concurrent.ExecutionException: java.lang.ClassCastException: class java.util.ArrayList cannot be cast to class org.diylc.announcements.Announcement (java.util.ArrayList is in module java.base of loader 'bootstrap'; org.diylc.announcements.Announcement is in unnamed module of loader 'app')
	at java.base/java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.base/java.util.concurrent.FutureTask.get(FutureTask.java:191)
	at java.desktop/javax.swing.SwingWorker.get(SwingWorker.java:613)
	at org.diylc.swing.gui.MainFrame$6.done(Unknown Source)
	at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750)
	at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:847)
	at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112)
	at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:857)
	at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:317)
	at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:249)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

Could this introduce the behaviour?

bancika, 2025-02-10:
The warning and the error are okay, it's unrelated.

Another option to try instead of what I sent previously is

-Dsun.java2d.uiScale=1.0

bancika, 2025-02-12:
Actually, in your case it should probably be

-Dsun.java2d.uiScale=2.0

bancika, 2025-02-13:
@janneman001  another possible solution I found is to set env variable and see if that helps

export GDK_SCALE=2

bancika, 2025-03-10:
@janneman001 did you have a chance to try 4.41 or newer version? There's a good chance thay your issue has been fixed

bancika, 2025-03-26:
closing due to inactivity, please reopen if still an issue

closed 2025-03-26 by bancika

sagebind, 2025-05-28:
For what its worth, I have the same issue. My system is running GNOME 48. Tested with DIYLC 5.5.0 AppImage:

[image]

This likely is Swing's fault somehow doing a bad job with HiDPI. For example, VisualVM which also uses Swing has the same issue: https://github.com/oracle/visualvm/issues/575. I tried all the recommendations, but fundamentally the issue is that the font size is always double what the UI scale is, regardless. E.g. a UI scale of 2.0 seems to result in a font scale of 4.0:

[image]