bancika/diy-layout-creator #880: Improve solder pad component

bancika, 2025-02-20:
- more options for SMD
- rectangular shape
- add angle of rotation
- backwards compatible

michaeleschuler, 2025-07-07:
Thanks for your reply! I am actually fairly new at this and have been using DIYLC to create vero layouts for pedal PCBs, but I recently started playing with a design that uses two layers to facilitate a compact PCB, since I saw that was possible, and so have been noticing some issues when exporting Gerber files. I have started to work with Eagle, but it is overkill for what I do, while DIYLC is nice in that it is much less complicated. The option to do the most basic type of two-layer board in DIYLC would be great, but I understand that this complicates things,  I don't know that I have the technical background to provide any good answers, given that I am fairly new to this, but here are a few thoughts.

1. Regarding vias: For a two layer board, having the option under pad creation to have a second identical pad automatically added on the bottom layer (and for these two pads to be grouped in some fashion so they move together, and for through hole plating to be turned on automatically) would seem to be the simplest solution, since my understanding is that a via should have some sort of annular ring on both ends of the through plating, Or maybe a via could be a separate component that is essentially two pads with through plating always on and a top and bottom layer identified.  These solutions don't address the use of inner layers (which seem to complicate things significantly) but would address the issue of having to manually create pads that are on top of one another visually but assigned to different PCB layers. 
2. Since DIYLC has the option of showing the underside of the board currently, having the option to have that display only traces and pads assigned to the bottom PCB layer, instead of a mirror image of the top layer, would be one option, as it allows the user to visually confirm that the back appears as expected.
3. Visually differentiating PCB layers does seems difficult given how layers are currently handled. Since (I think) only traces, ground fill, and pads can be added to other PCB layers, maybe add an option under the Layers menu to change trace visibility and locking separately for each PCB layer? For example, allow "Trace" to be replaced with separate "Trace, Top" and "Trace, Bottom" options. This would allow bottom traces to be dimmed, which would be a workaround for visually differentiating them, or allow everything except bottom layer traces to be hidden, which resolves the issue under #2, above.
4. If separate vias (i.e,, two pads joined by through plating) are an option, visually differentiating pads with through plating doesn't seem necessary, since the default appearance of regular pads and vias could be different.

In any case, I've found DIYLC invaluable as it is, so I don't want to request anything that would create problems.

bancika, 2025-12-29:
@ronnimax can't think of an easy way to do it

bancika, 2026-11-12:
@ronnimax 6.11 will add support for vias that connect top and bottom copper surfaces. Also PTH pads will produce a pad on both sides of the board and the hole is plated through. Similar result to via but a bit different as you can solder through that hole unlike the via.