bancika/diy-layout-creator #885: Newly added AppImage can't be launched, possibly due to missing AppRun file

hnjh537, 2025-03-07:
Hi, first of all, thank you for your awesome work.

I've tried to launch the Linux AppImage that was included in the v4.41.0 Release.
However, it returns the following error: `execv error: No such file or directory`

Steps to reproduce:
- Download and unzip
- Make executable: `$ chmod +x DIYLayoutCreator-4.41.0-x86_64.AppImage `
- Run AppImage: `$ ./DIYLayoutCreator-4.41.0-x86_64.AppImage`
- Output: `execv error: No such file or directory`

Possible issue:

- Extract AppImage: `$ ./DIYLayoutCreator-4.41.0-x86_64.AppImage --appimage-extract`
- `$ cd squashfs-root`
- `$ fd AppRun` does not return any results

I suspect the error occurs because the AppRun file is missing. Since I don't have much experience with AppImages, this might not be the root cause.

ahakkoca07, 2025-03-07:
I have the same problem, it doesn't launch in any way in Linux.

hnjh537, 2025-03-07:
> I have the same problem, it doesn't launch in any way in Linux.

If you download this zip and run the run.sh file, you can still use the application: [diylc-4.41.0.zip](https://github.com/bancika/diy-layout-creator/releases/download/v4.41.0/diylc-4.41.0.zip)

bancika, 2025-03-07:
Thanks for reporting. I don't have a linux to try and just learned about appimage, so it will be a process :) I'll build a new version and upload it for testing

bancika, 2025-03-07:
@ahakkoca07 @hnjh537 please try 4.41.1 that I just published

ahakkoca07, 2025-03-07:
Still same for me:
```
ahakkoca07@mxlinux:~/Downloads
$ ./DIYLayoutCreator-4.41.1-x86_64.AppImage
execv error: No such file or directory
```

bancika, 2025-03-07:
Finally found a way to test under windows and got it to work! @ahakkoca07 @hnjh537 please try 4.41.1 again

ahakkoca07, 2025-03-09:
It works! Thanks @bancika

bancika, 2025-03-09:
Cool. Thanks for the update

closed 2025-03-09 by bancika