bancika/diy-layout-creator #893: v4.41.1  & v5.0 osx-arm.... laggy & cluncky. ??

G-C67, 2025-03-16:
Hi, 
I'm new here and would first of all like to say Hi to everyone.  

I have been using V4.40. osx-arm for a while and just upgraded to V4.41, then v5.0. 
I immediately noticed that the response of the newer versions feels delayed and clunky in comparison to V4.40. 
Using on MacBook Air M1. I have tried deleting and reinstalling but getting the same slow responses. 
Went back to v4.40 osx-arm and the response and functionality is smooth and fast. 

Has anyone else experienced this ?

Thanks

Gaz.

bancika, 2025-03-17:
Did you download the arm version of diylc? Can you check the activity
monitor and ensure that the app is running in silicon mode?

bancika, 2025-03-17:
I was able to replicate the slowness...will investigate deeper. Thanks

G-C67, 2025-03-17:
Thanks for the quick reply. 
I have run Vs5.0 again and checked activity monitor .
Does show as running ARM64. 

[image]

G-C67, 2025-03-17:
Checked V4.40
and activity monitor looks similar, but means nothing to me..

[image]

bancika, 2025-03-17:
Thanks for the info. I found one bug that I will fix soon. Also this
version introduces high DPI support for high resolution screens. Images
render at higher resolution internally in order to be smoother. Rendering
looks smoother but inevitably it makes things slower. I will make it
configurable so you can turn it off.

bancika, 2025-03-17:
> Great.
> Love the application.
> So do both V4.41 and v5.0 use the high DPI ?

Yes, the changelog shows that the support was added in 4.41. Also I
introduced a bug in caching which affects performance :)

bancika, 2025-03-17:
In the meantime you can also try to hide rulers from the Config menu, they
slow down things considerably

G-C67, 2025-03-18:
I tried turning off rulers, didn't make much difference. 

But I did turn off "Enable Cache Boost"... and that made a big difference. Faster and smoother and seems as smooth and responsive as v4.40, if not better. Even in hi-quality rendering which is something I don't usually bother with.
Great work. Using v5.00 now (with Cache Boost off), DPI increase looks nice and sharp.
V4.41.1 seems to work great also with Cash boost off.

bancika, 2025-03-18:
Great, I think you will be pleased with 5.1.0, it flies :)

bancika, 2025-03-19:
closing, pending release