bancika/diy-layout-creator #894: v5.0.0 graphics look wrong (pixelated, maybe scaling issue?)

blackcorvo, 2025-03-17:
I'm on Windows 10.

For comparison, here's v4.40.0 (that I still have installed on 'Program Files (x86)' )

[image]

and here's v5.0.0

[image]

I literally just updated it and immediately noticed the strange graphics.

bancika, 2025-03-17:
Do you have a high dpi monitor with pixel scaling? Is anything else
affected other than icons in the toolbox? The drawing itself should be fine

blackcorvo, 2025-03-17:
It's a 1920x1080 screen, with text scaling at 125%.
Even the startup animation has been affected:

[image]

And weirdly, at the same time that the text and graphics seem to have been scaled down in v5, the workspace seems to be zoomed in now. 75% zoom in v5 looks like 100% zoom in V4.

V5
[image]

V4
[image]

bancika, 2025-03-17:
V4.40 and previous didn't honor scaling factor, so it's expected that it will look bigger if your display has scaling configured.

blackcorvo, 2025-03-18:
> V4.40 and previous didn't honor scaling factor, so it's expected that it will look bigger in the new version if your display has scaling configured.

Would it be possible to add a toggable option for old scaling in a future update? Because even if the jagged scaling wasn't there, having to scale the project to 75% every time just so I can see the entire thing is simply very annoying, and I'd rather not have to compromise readability on all other apps just to use this one - which would also be hard to read because the text at 100% is absolutely miniscule on my screen:

[image]

bancika, 2025-03-18:
Yes, it will be configurable

bancika, 2025-03-18:
Although I'm not sure if it would help in your case because you are scaling
your display to 125% and everything SHOULD be bigger, right?

blackcorvo, 2025-03-19:
Just a little bit, but yes.

V4 @ 100%

[image]

V4 @ 125%

[image]

But it's enough of a difference for the scaled-up version to severely reduce eye strain for me - and back pain too, since I don't need to shimp towards the screen to read.

bancika, 2025-03-26:
does 5.1.0 work better?

pedroberretta, 2025-03-26:
> does 5.1.0 work better?

On my computer, if I set both "High DPI Rendering" and "Hi-Quality Rendering", the workspace area looks ok, but the icons in the menu and the drawing at the start still look poor quality. In v4.40.0 all graphics looks greate.

[image]

[image]

bancika, 2025-03-26:
@pedroberretta what is your scale factor of the display?

pedroberretta, 2025-03-26:
> [@pedroberretta](https://github.com/pedroberretta) what is your scale factor of the display?

My display scale factor is set to 125%, which is the recommended setting by the system.

shanemneal, 2025-03-27:
Same thing happening to me with 5.1.1. Everything looks perfectly fine at 100%, but the system recommends 125% and if I run scaling at 125%  per recommended default, everything becomes all pixelated. I don't mean to be annoying since the developer is aware but figured I'd mention it here in case anyone finds a fix.

blackcorvo, 2025-04-02:
> does 5.1.0 work better?

No, still the same.

[image]

Also, I'd like to add that I tried v5 and up on my work PC (where I can only use the zipped versions, as I have no install priviledges), and it throws an error as if I didn't have JAVA installed. Meanwhile I have no issues with v4.41.1. And yes, I've tried just copying the new files over into a copy of the app's folder, same error.

Sorry for throwing all these issues at you, I just wanted you to be aware of them. I'll be sticking to v4.41.1 for the time being, works fine for what I need.

bancika, 2025-04-02:
did you try 5.1.1 that failed or 5.1.0? I fixed windows deployment with
5.1.0

blackcorvo, 2025-04-02:
That screenshot is from v5.1.1 on my home laptop. Changing rendering options does not affect the menus, only the work area.

On my work pc (HP Touchsmart 420 all-in-one w/ Windows 10), ONLY v4.40.0 works, and I have to use the zip file version because of not having install privileges on this machine.

• v4.41.0-v5.0.0 from the zip file gives me the error "A Java Exception has occurred".
I've noticed all of these versions lack an exe file when extracted;

• v5.1.1 from the zip file gives me the error "This application requires a Java Runtime Environment 17 (16-bit)".
This version's zip file has the jar and exe files, but only the jre folder.

bancika, 2025-12-26:
@blackcorvo can you please confirm that this is no longer issue in more recent versions? I renamed jre to jre17 in the deployment

closed 2025-12-26 by blackcorvo