bancika/diy-layout-creator #897: Cant undo when dragging components with mouse/trackpad

UrantiaElectronics, 2025-03-23:
Since I updated to the last version, I'm having this strange issue.
Whe I drag a component with the mouse or my trackpad (doesn't matter if it's the whole component or the pins on the terminals), can't undo this action.
I I use "ctrl+Q" or if I move the component with the arrowns in my keyboard everything works fine, it's just when I drag using the mouse.
The problem gets worse if I do something like adding a component, then moving a few ones, When I ctrl+z (or undo in the menu) the project goest to the point whe I added the last component so the app somehow is registering everithing but the components moves done with the mouse or trackpad.

I tried to reinstall the app, deleting my config.xml, reinstalling java from scratch, etc. with no success. I'm using Windows 11 (updated).

Do you know how can it be fixed? I'm assuming it's a problem with my setup, not DIYLC itself but I'm not a programmer or have enough knowledge to fix this ;(

Thanks in advance, and of course thanls a lot for this wonderfull software :)

UrantiaElectronics, 2025-03-23:
just in case, here's my latest log file:

12:15:52,909  INFO [main] ConfigurationManager - Initializing configuration for: diylc
12:15:53,818  INFO [main] ConfigurationManager - Saving configuration
12:15:54,719  INFO [main] DIYLCStarter - JarLoader strategy: agent
12:15:54,719  INFO [main] DIYLCStarter - Loading languages...
12:15:54,723  INFO [main] LangUtil - Loading translation for English
12:15:54,726 DEBUG [main] LangUtil - Loading language file: C:\Program Files\DIYLC\lang\English.txt
12:15:54,747 DEBUG [main] DIYLCStarter - Java version: 17.0.14+7 by Eclipse Adoptium
12:15:54,748 DEBUG [main] DIYLCStarter - Java home: C:\Program Files\DIYLC\jre17
12:15:54,749 DEBUG [main] DIYLCStarter - OS: Windows 11 10.0
12:15:54,750  INFO [main] DIYLCStarter - Starting DIYLC with working directory C:\Program Files\DIYLC
12:15:54,784  INFO [main] DIYLCStarter - Configuration dump start.
12:15:54,786  INFO [main] DIYLCStarter - antiAliasing = true
12:15:54,786  INFO [main] DIYLCStarter - hiQualityRender = false
12:15:54,786  INFO [main] DIYLCStarter - continuousCreation = true
12:15:54,786  INFO [main] DIYLCStarter - autoEdit = false
12:15:54,787  INFO [main] DIYLCStarter - outline = false
12:15:54,787  INFO [main] DIYLCStarter - highlightContinuityArea = false
12:15:54,787  INFO [main] DIYLCStarter - extraSpace = true
12:15:54,788  INFO [main] DIYLCStarter - cachingEnabled = true
12:15:54,788  INFO [main] DIYLCStarter - defaultBlocksImported = true
12:15:54,789  INFO [main] DIYLCStarter - Configuration dump end.
12:15:54,856  INFO [main] DIYLCStarter - Creating the main frame...
12:15:54,861  INFO [Thread-1] FontOptimizer - Starting font optimizer
12:15:58,089 DEBUG [Thread-1] FontOptimizer - Fonts loaded in 0ms
12:15:58,384  INFO [main] Presenter - Current DIYLC version: 5.0.0
12:15:58,418  INFO [main] Presenter - Loading component types.
12:15:58,421 DEBUG [main] Utils - Scanning jar: /C:/Program Files/DIYLC/lib/diylc-core.jar
12:15:58,460 DEBUG [main] Utils - Scanning jar: /C:/Program Files/DIYLC/library/main.jar
12:15:58,885  INFO [main] Presenter - Loading additional library JARs
12:15:58,888 DEBUG [main] JarScanner - Found JAR file: main.jar
12:15:58,888 DEBUG [main] JarScanner - Scanning main.jar
12:15:58,950  INFO [main] ClassLoaderUtil - Adding file to the classpath: C:\Program Files\DIYLC\library\main.jar
12:15:58,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.Abstract3LegSymbol
12:15:58,955 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractLeadedComponent
12:15:58,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractMultiPartComponent
12:15:58,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.AbstractSchematicLeadedSymbol
12:15:58,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractBoard
12:15:58,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractProtoBoard
12:15:58,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.AbstractVeroBoard
12:15:58,956 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.BlankBoard
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.Breadboard
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.EurorackStripboard
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.EyeletBoard
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.MarshallPerfBoard
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.PerfBoard
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.ProtoBoard
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.ProtoBoard780
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TagStrip
12:15:58,957 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TerminalStrip
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.TriPadBoard
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.boards.VeroBoard
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.ChassisPanel
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.DialScale
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.EllipticalCutout
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.PolygonalCutout
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.chassis.RectangularCutout
12:15:58,958 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CopperTrace
12:15:58,959 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CurvedTrace
12:15:58,959 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.CutLine
12:15:58,959 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Dot
12:15:58,959 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Eyelet
12:15:58,959 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.GroundFill
12:15:58,959 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.HookupWire
12:15:58,959 DEBUG [main] JarScanner - Loaded class: org.diylc.components.connectivity.Jumper
[... cut ...]
12:24:04,284 DEBUG [AWT-EventQueue-0] Presenter - dragStarted(java.awt.Point[x=1413,y=1264], 2)
12:24:04,467 DEBUG [AWT-EventQueue-0] Presenter - dragEnded(java.awt.Point[x=1421,y=1264])
12:24:04,561 DEBUG [AWT-EventQueue-0] Presenter - dragEnded(null)
12:24:06,300 DEBUG [AWT-EventQueue-0] Presenter - dragStarted(java.awt.Point[x=1418,y=1257], 2)
12:24:06,834 DEBUG [AWT-EventQueue-0] Presenter - dragEnded(java.awt.Point[x=1433,y=1254])
12:24:06,927 DEBUG [AWT-EventQueue-0] Presenter - dragEnded(null)
12:24:14,564  INFO [AWT-EventQueue-0] MainFrame - Closing the app

bancika, 2025-03-23:
Fixed for 5.1.0

closed 2025-03-26 by bancika