bancika/diy-layout-creator #899: No icons in universal zip for 5.1?

alandmoore, 2025-03-26:
Hello, I maintain the Diylc AUR package for arch linux.  In previous releases the generic zip (which I assume is now the universal zip) contained icon files, which our install script used.  Is there a reason these were dropped from the universal zip in 5.1.0, and would it be a problem for you to continue including them in future releases?

Thanks for you help.

alandmoore, 2025-03-26:
Thanks!  We can always add them manually, but of course if you were to rebrand in any way we'd want to stay in sync with that.

bancika, 2025-03-26:
Interestingly, I didn't even know about the AUR package and arch linux :)

closed 2025-03-26 by bancika

alandmoore, 2025-03-26:
There is a small but devoted group of us using DIYLC on Arch.  Not sure exactly how many, but I definitely get feedback when I muck something up :).