bancika/diy-layout-creator #900: DIYLC 5.1.0 Fails to launch on Windows 11

shanemneal, 2025-03-27:
First and foremost, thank you for everything!

Anywho just recently updated from 4.7 which was working fine to 5.1 and after multiple restarts and ensuring 4.7 was removed, I noticed that 5.1 fails to launch on Windows 11 on my personal laptop, even when I try running as admin. My system specs are as follows:

Edition	Windows 11 Home
Version	24H2
Installed on	‎2/‎10/‎2025
OS build	26100.3476
Experience	Windows Feature Experience Pack 1000.26100.54.0

Thanks!

bancika, 2025-03-27:
What does it mean exactly? Nothing happens or there's an error? Is there
anything in logs? You can find them in c:/users/you/diylc/logs

shanemneal, 2025-03-27:
Apologies on the lack of info, nothing honestly seems to happen when I try to use the program. I will click on the icon for it but nothing seems to happen. Let me try again, and attach any logs to this

Thank you again!

shanemneal, 2025-03-27:
Hey again,

Attached are the logs 

[log-2025-03-27 10.35.06.txt](https://github.com/user-attachments/files/19489209/log-2025-03-27.10.35.06.txt)
[log-2025-03-27 10.33.15.txt](https://github.com/user-attachments/files/19489210/log-2025-03-27.10.33.15.txt)

bancika, 2025-03-27:
Seems like you are running an old version, logs say Current DIYLC version:
4.37.0

Please try to delete c:\Program files (x86)\DIYLC and install again

shanemneal, 2025-03-27:
Just deleted it manually and also selected uninstall from program files.  I reinstalled and tried again, but the same issue seems to persist. I went to the same folder to see if there were any logs, but there didnt appear to be any updated logs or anything.

TK4US, 2025-03-27:
I confirm, I meet the exact same problem. I uninstalled/reinstalled DIYLC 5.1 with no success, nothing happens when clicking the icon. I don't have any previous version installed.
I was previously running V4 flawlessly, always under W11.

shanemneal, 2025-03-27:
Here's a log just in case:

[log-2025-03-27 02.42.53.txt](https://github.com/user-attachments/files/19492364/log-2025-03-27.02.42.53.txt)

bancika, 2025-03-27:
That's totally unrelated. Not sure why it happens, it works fine on my mac
with x2 scaling.

shanemneal, 2025-03-27:
Gotcha, yeah windows is pretty dumb sometimes lol. I'll close this since it's unrelated. I appreciate it though!

closed 2025-03-27 by shanemneal

TK4US, 2025-03-28:
Thank you, it works perfectly now 👏
Nice work for so many years now !