bancika/diy-layout-creator #901: DIYLC v5.1.0 not starting on Linux Mint 22.1

greizdeifi, 2025-03-30:
Hi 
first, thanks for the great work!

The new release v5.1.0 is not starting up on Linux Mint 22.1. It does not do anything when I execute the AppImage. v5.0.0 works fine. As an observation I've noted that the v5.1.0 image file (DIYLayoutCreator-5.1.0-x86_64.AppImage, 21 MB)  is 35 MB smaller than v5.0.0 (DIYLayoutCreator-5.0.0-x86_64.AppImage, 56 MB).

Thanks,
Michael

bancika, 2025-03-30:
Just install java jre 17 or later. The new version doesn't bundle it.m.I
had some issues with that

greizdeifi, 2025-03-30:
Oh, I was not aware of that. Just installed java - issue solved. Works fine now. Thank you very much for the fast reply!

closed 2025-03-30 by greizdeifi

bancika, 2025-03-30:
I forgot to put it in the notes...hoping still that I can bundle it back

bancika, 2025-05-27:
> I have the same issue with every version since v5.1.0, even though I've
> installed the latest OpenJDK 24 runtime.
>
> OS: Nobara Linux (Fedora-based) / KDE Plasma / Wayland
>
> When I run
> ./DIYLayoutCreator-5.5.0-x86_64_435f89d1f82617dea914a4a979680710.AppImage
> , I get this:
>
> /usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
> AppRun: Running from directory: /tmp/.mount_DIYLayEYX0Xs
> AppRun: Executing /tmp/.mount_DIYLayEYX0Xs/usr/bin/diylc.sh with args:
> diylc.sh: Running from directory: /tmp/.mount_DIYLayEYX0Xs/usr/bin
> diylc.sh: Using system Java: /usr/bin/java
> diylc.sh: Current working directory: /tmp/.mount_DIYLayEYX0Xs/usr/bin
> diylc.sh: Executing DIYLCStarter with Java
> Splash screen could not be initialized:
> No X11 DISPLAY variable was set,
> or no headful library support was found,
> but this program performed an operation which requires it,
>
> 15:11:13,503  INFO [main] ConfigurationManager - Initializing configuration for: diylc
> 15:11:13,896  INFO [main] DIYLCStarter - Loading languages...
> 15:11:13,897  INFO [main] LangUtil - Loading translation for English
> 15:11:13,903 DEBUG [main] DIYLCStarter - Java version: 21.0.7+6 by Red Hat, Inc.
> [... 33 lines cut ...]
>         at org.diylc.DIYLCStarter.main(DIYLCStarter.java:65)
> 15:11:15,095 DEBUG [Thread-0] FontOptimizer - Fonts loaded in 0ms
> 15:11:15,870  INFO [Thread-0] FontOptimizer - Font optimization completed in 745ms. Found 6 slow fonts
> 15:11:15,871 DEBUG [Thread-0] FontOptimizer - Slow fonts: Adwaita Mono,Noto Sans CJK TC,Noto Serif SemiBold,Noto Sans CJK JP,Noto Serif SemiCondensed ExtraLight,C059
> 15:11:15,871  INFO [Thread-0] ConfigurationManager - Saving configuration

it's something about your java. Try with this

sudo apt install openjdk-17-jre

ErebosGR, 2025-05-27:
I found what the issue was.
The `java-latest-openjdk` meta package installed 2 headless versions (21 and 24). I uninstalled it and installed only 21, and now it works.
Thank you.

bancika, 2025-05-27:
cool