bancika/diy-layout-creator #902: How to use underside rendering

awenisko, 2025-03-30:
Hi, there's a mention about underside rendering in the new release notes, but I don't seem to be able to find this feature anywhere in the GUI. Where it is?

bancika, 2025-03-30:
Double click on a board and select direction and offset of underside display

awenisko, 2025-03-30:
Found it, thank you. However:
- the trace cuts are barely visible in the underside render. Locating the trace cuts is the main use-case I plan to use this feature for.
- SIP IC component has some bug - when placed outside of the board and then moved in, the position of the component jumps from the main board to the underside render.
Anyway, great feature!

awenisko, 2025-03-30:
Another (low severity) issue is with the Potentiometer component. It has "Shaft Up" and "Shaft Down" views, but the underside render does not reflect that and the rendering of the potentiometer is the same.

bancika, 2025-05-03:
Fixed the bug with SIP for 5.4.0

closed 2025-05-18 by bancika