bancika/diy-layout-creator #903: [Question] Grouping with hierarchy (parents, childs, etc...)

Kirbyrawr, 2025-03-30:
Hi there, is it possible to have parents and childs in the group?
If it's not possible, is it in a roadmap or feature that will be included?

Thanks in advance.

[image]

bancika, 2025-03-30:
I did not have it in mind earlier. What would be the benefit of such feature?
I usually implement features based on the impact, complexity, affected user base, etc. If you really want something, you can always sponsor a feature a push it up the queue.

Kirbyrawr, 2025-03-31:
Hi there, thanks for the reply.
The idea is for the building blocks to be contained as a group, for example an ESP32 you can name the GPIO etc... and have everything contained there.