bancika/diy-layout-creator #904: 4 way switch?

knight-yyz, 2025-03-30:
Any chance of adding a 4 way switch to the mix?  just needs 2 more lugs

bancika, 2025-03-30:
What kind of the switch exactly?

knight-yyz, 2025-03-30:
4 way lever switch,  same as the 3 way or 5 way levers,  but there is an extra lug on each row

[image]

bancika, 2025-03-30:
There should be 4 way lever switches already. Did you try double clicking
on a lever switch?

knight-yyz, 2025-03-30:
not in gen 5 ,  just 5 way and super

bancika, 2025-03-30:
it's definitely there

[image]

closed 2025-03-30 by bancika

knight-yyz, 2025-03-30:
how did you do that?

bancika, 2025-03-30:
I mentioned it in the previous message - double click on a lever switch
(once it's on the drawing). I suggest reading the manual, there's some
useful info, although should admittedly add more info
https://github.com/bancika/diy-layout-creator/blob/wiki/Manual.md

knight-yyz, 2025-03-30:
you didnt say once it was in the drawing