bancika/diy-layout-creator #907: BJT - text and pin layout

UbuntuRikard, 2025-04-04:
If I could wish for a small change to DIYLC, it would be for transistors.
I could imagine being able to choose which pins are C, E, and B; by myself. The combinations available are not entirely comprehensive. It would also make sense if one could choose what is displayed when placing a transistor, such as Q1, BC547, B-C-E.
It would also be smart if one could click on a point in the properties to move the associated text for a transistor and then lock it again.

bancika, 2025-04-08:
What are the combinations that you miss?

UbuntuRikard, 2025-04-08:
I might have been to hastily sorry, can't find a case right out of the box, but it's more the possibility to see pinout the Q number and the name and the transistor: BC547
The option for selecting the pinout freely might not be so important, as i think. But the above, really a wish

bancika, 2025-04-08:
I understand, but I find that using an additional label component (or two)
to add whatever info you may need is a flexible and good enough solution.
Transistors are small and automatically labeling them is a pain, so I
didn't want to reinvent the wheel since there is a Label component that can
display any text you need

bancika, 2025-12-28:
Closing this as it can easily be achieved with the existing Label component and potentially grouping together

closed 2025-12-28 by bancika