bancika/diy-layout-creator #909: DP3T unable to connect pins 2, 6 and 8

ofesad, 2025-04-07:
Whatever I try, I cant connect pins 2, 6 and 8.
Moving the switch shows the connected ones but I cant hook any other cables to the empty pins.

[image]

bancika, 2025-04-07:
I don't follow. Please provide clear instructions

ofesad, 2025-04-07:
To clarify:
For some reason, it's impossible to hook up the pins n° 2,6 and 8 with the hookup wire. All other pins work fine.
Happens in any orientation or position. Tested with new project.

bancika, 2025-04-07:
The wire doesn't stick or it's impossible to move the endpoint to the
terminal? What kind of switch?

ofesad, 2025-04-07:
The wire doesnt stick. It will stick to any other pin except the mentioned.
Lever Switch DP3T

bancika, 2025-04-14:
Please try the latest version, I have verified that it works fine, as much as I move the switch, all the wires stay in place

[image]

ofesad, 2025-04-26:
[Updated to ver 5.30]
Problem persist. 
Just to confirm, it's the DP3T (import 5 position)

[image]

bancika, 2025-04-26:
I am not sure I 100% understand the issue. Maybe you just need to uncheck
snap to grid from the config?

ofesad, 2025-04-26:
THAT WAS IT! Didnt notice it because every other component would snap/connect just fine!
Gosh.
Sorry for bothering so much with nothing.

closed 2025-04-26 by ofesad