bancika/diy-layout-creator #911: Config Language

OliveZeFrenechi, 2025-04-10:
Hi,

I just downloaded V5.11, on two Windows 11 laptops and this feature doesn't work.
I tried to switch to French, and the tool, even after a reboot, still displays in English.

I do had a look into the config.xml file and the parameter seems to be updated correctly

Olivier

bancika, 2025-04-10:
is there any error in the log?

OliveZeFrenechi, 2025-04-10:
### 16:00:02,857  INFO [main] ConfigurationManager - Initializing configuration for: diylc
16:00:04,130  INFO [main] ConfigurationManager - Saving configuration
16:00:04,757  INFO [main] DIYLCStarter - Loading languages...
16:00:04,764  INFO [main] LangUtil - Loading translation for Français
16:00:04,786 ERROR [main] LangUtil - Error loading language file
java.lang.NullPointerException
	at java.base/java.io.Reader.<init>(Unknown Source)
	at java.base/java.io.InputStreamReader.<init>(Unknown Source)
	at org.diylc.lang.LangUtil.load(LangUtil.java:93)
	at org.diylc.lang.LangUtil.configure(LangUtil.java:85)
	at org.diylc.DIYLCStarter.runDIYLC(DIYLCStarter.java:109)
	at org.diylc.DIYLCStarter.main(DIYLCStarter.java:66)

closed 2025-04-17 by bancika