bancika/diy-layout-creator #913: Custom prototype boards?

Miq1, 2025-04-18:
I just found this tool and now am trying to design a circuit on one of the notorious green prototype boards you will find all over the net, particularly on this:

[image]

It has the 22x27 pads all set as vias.

In the boards list I find the "Proto Board 780 Holes" that comes close, but is too large. Can I cut that somehow to fit the desired dimensions?

bancika, 2025-04-20:
You cannot change the existing component unless you go in the codebase and make actual code changes.
One way to build this board is to just create it using basic components, like pads and blank board and then save it as a building block. You can also leave this here as a request and I will eventually do it, but there are no guarantees.
Also, you can sponsor a component and have it implemented sooner.

Cheers

Miq1, 2025-04-21:
Hi @bancika , thank you for considering. I helped myself with the second advice you gave and used pads for the soldering points I was going to use.
Once I got the hang of it it was quite easy in fact.

Thank you for this useful piece of software! ❤️

closed 2025-04-21 by Miq1