bancika/diy-layout-creator #915: Mouseover colors for tooltips

Schaagi, 2025-04-24:
When I move the mouse over an icon, a text is displayed. The text is white, the background is light yellow, so it is not legible. Can I adjust this somewhere?

bancika, 2025-04-24:
which icon?

Schaagi, 2025-04-24:
Concerns all icons for example these on top right and bottom right.

bancika, 2025-04-24:
do you have a screenshot? it's black against yellow on mine

Schaagi, 2025-04-24:
[image]

Hopefully you can see the screenshot (is my first) :-)

bancika, 2025-04-25:
Must be something about the system or L&F...which OS is that?

Schaagi, 2025-04-25:
Linux Mint 22.1

Schaagi, 2025-04-25:
I have tested it with other applications / programs. The yellow background is the same, but the text is black instead of white. 
I can only find the white tooltip text in your program.

Schaagi, 2025-05-16:
This problem with the unreadable tooltips only exists in Linux, everything is ok in Windows.

closed 2025-12-26 by bancika