bancika/diy-layout-creator #916: Feature Request: Ibanez guitar pickup selector 2502N

ofesad, 2025-04-27:
Hi. I have 2 Ibanez guitars: RG652 (HH) and RG657 (HSH).

I had no problem when replicating the RG657 circuit. Also I uploaded it to the cloud, so anyone can use it or check it.

However the Rg652 uses a different pickup selector, known as the 2502N:

[image]

Heres' the pickup operation chart:
[image]

Thank you in advance for your hard work!

Ofesad

ofesad, 2025-04-27:
[IBANEZ RG652 HH DIAGRAM.zip](https://github.com/user-attachments/files/19930913/IBANEZ.RG652.HH.DIAGRAM.zip)

Results should be (same as stated on the Dimarzio paper):
Pos.1 = Neck humbucker series
Pos.2 = Neck humbucker paralell
Pos.3 = Neck + Bridge 
Pos.4 = Neck inner + Bridge inner
Pos.5 = Bridge humbucker series

Using the 5 way switch (DP3T - Import 5-position strat) will not give that result, since it's a common 5-way.

The 5-way 2502N pickup selector is different to the typical 5 way pickup selector.

[image]

bancika, 2025-04-27:
Hm, if I understand correctly you are expecting a simple 5 way switch to
work like 2502N? I don't think you can achieve that

ofesad, 2025-04-27:
Of course not. They are different selectors.
What I am requesting is to add the Ibanez 2502N as a component, if possible.
That's all.

bancika, 2025-04-27:
Ok, got it...so there's no issue, just a new feature. It's possible to add,
of course. I can do it next time I do something around guitar wiring
components. There's also an option to sponsor a new feature and expedite
the development and get it in the next release.

maratbakirov, 2025-05-11:
https://www.youtube.com/watch?app=desktop&v=W08wTuX_ny4

this might be helpfull

StateOfEpicicity, 2025-06-15:
I just noticed this request; very cool!  It's an interesting switch; that's actually what came with my Schecter Sun Valley Super Shredder FR Exotic, and it allows a cool configuration.  It took me a bit of searching to figure out this is a common Ibanez switch, and I was surprised to learn that.  I'm using a standard 5-way superswitch at the moment, but if it is being installed at the factory on some Schecter models too, I wonder how ubiquitous it actually is!

downrazor11, 2025-12-16:
I'm not sure if this is still needed or helpful, but I have labeled some info on the switch with a detailed pic of the pcb exposed here:
[https://www.tdpri.com/threads/custom-wiring-s-h-with-2502n-5-way-switch.1146021/](https://www.tdpri.com/threads/custom-wiring-s-h-with-2502n-5-way-switch.1146021/)

[2502N Map](https://atherispickguards.wordpress.com/wp-content/uploads/2019/06/ibanez3ps1sc5_cortek2502n_map.png)

mm-rnd, 2025-12-16:
The map diagram you just added shows bridge as pos1 but the info further up has neck pos1.

Anyway, I updated the component connections to work as I would expect, as detailed on this diagram:
https://guitarpartscenter.eu/en_US/p/ALPHA-SW-2502N-5-Way-Lever-Switch-ADWH/23036

You can see the result of the analysis. I've no idea when it will make it into the codebase and a release, I have several pull requests pending. But I did the wiring as per the diagram in the link and all positions check out.
[2502N Updated.html](https://github.com/user-attachments/files/24200671/2502N.Updated.html)

downrazor11, 2025-12-16:
Keep in mind or be aware of the orientation of the lugs on the map of the switch and compare them to the pics below, as long as the Thomann one is made identical to the Ibanez (Cor-Tek) one, the map is accurate.

Here's a 2502N switch wired into one of my strat harnesses (top-down view of the switch/pcb):
[2502N Strat Harness Top-down](https://atherispickguards.wordpress.com/wp-content/uploads/2025/12/pxl_20251216_222512625.jpg)

Here's the backside (2502N-labeled side of the switch):
[2502N Strat Harness Backside](https://atherispickguards.wordpress.com/wp-content/uploads/2025/12/pxl_20251216_222443918.jpg)

And here is the PCB that I made for the switch:
[Atheris 2502N quickconnect PCB](https://atherispickguards.wordpress.com/wp-content/uploads/2025/12/atheris_oshpark_swbrd_importsw_slot_5way2502nb1_041822.png)