bancika/diy-layout-creator #920: Windows Security popup when executing installer or executable

cedricus, 2025-05-02:
Hello,

I found your software and I can't wait to try it.

I dowloaded the last version: diylc-5.3.0-win64.exe but when opening, Windows pop up "Windows Defender protected your computer" and blocks the installation.

So I tried with the zip package. But when launching diylc.exe I have the same security alert.

Any way to secure the installer to avoid this kind of alert?

Thank you.

bancika, 2025-05-03:
Hi. It's a false positive error caused by the installer. I'm not sure why
you get it with the app as well, but it's 100% safe as long as you
downloaded it from the official sites.

I'm exploring some options but there don't seem to be any free options
available. For mac users I'm forced to limit the signed installer to paid
patrons only because it costs me money (and effort) to notarize. I'd have
to apply a similar model for windows I'm afraid.

Cheers

cedricus, 2025-05-21:
Ok thank you.
Can't wait to test all that to create uitar wirin diagrams :-)

Thank you for providing this.
I will see how I can contribute.

closed 2025-05-21 by cedricus